birkschmithuesen/MergingEntities #12: Create all directions from Trainingdata

birkschmithuesen, 2022-06-27:
* to get rid of orientation, all the Quarternion data should be rotate by one degree at a time, or so, to create training data where the performer virtually rotate in all directions

danieldalfovo, 2023-03-11:
is this something on my end? if yes we need to discuss this since I don't know what needs to be done

birkschmithuesen, 2023-03-12:
Yes, hope that can be solved on your end. Have you checked my video messages for this? https://cloud.artesmobiles.art/index.php/f/2849087
I would be available for a call monday after 2.00pm

danieldalfovo, 2023-03-12:
cool. forgot about these. nice, perfect. how about tuesday for a call? tomorrow a bit busy.

danieldalfovo, 2023-03-12:
the idea to generate extra synthetic training data is interesting and may work. i think it could also be worth trying to not work with absolute degrees but with degree movement/changes, that would for example catch the case if "z" is not moving, but "y" is moving up and down, in any "z" orientation. so its rather learning a motion than a position.

system-failed, 2023-04-11:
we tried some ways, but manual training of different directions seems to be most promising

closed 2023-04-11 by system-failed