birkschmithuesen/MergingEntities #68: when a sensor is not connected, the OSC data flow freezes every 5 sec or so, because of an I2C error

birkschmithuesen, 2023-04-01:
sensor for left_upper_arm is not usable
-135.23// -96.45// 38.77// 21.25// 60.02// 38.77// 106.64// 145.41// 38.77// -9.00// 29.77// 38.77// 11.47// 50.24// 38.77// 
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
-134.18// -95.41// 38.77// 21.27// 60.04// 38.77// 106.40// 145.17// 38.77// -8.40// 30.37// 38.77// 11.45// 50.22// 38.77// 
trying to resurrect left_upper_arm
* setting up gyro ... failed
* loading calibration dataI2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
I2C ERROR CODE : 2
 ... worked
127.46// 166.23// 38.77// 21.00// 59.77// 38.77// 104.58// 143.35// 38.77// 3.63// 42.40// 38.77// 10.97// 49.74// 38.77// 
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
sensor for left_upper_arm is not usable
134.74// 173.51// 38.77// 21.27// 60.04// 38.77// 104.80// 143.57// 38.77// 1.90// 40.67// 38.77// 10.91// 49.68// 38.77// 
sensor for left_upper_arm is not usable

system-failed, 2023-04-02:
it works fine now, that there is no "freeze" in the data flow any more, but now when one sensor is disconnected, two sensor data flow stops. It seems that two inputs are somehow connected, when one fails physically both fail in the code

tpltnt, 2023-05-01:
not part of the current development tree. A new issue may be created it arises in the new firmware

commit a4d732c0 (2023-04-02): stop resurection early, closes #68
closed 2023-05-01 by tpltnt